eyal0/OctoPrint-TimeToFilament #2: Problem with operation

KenLucke, 2020-05-27:
I have installed this, and I seem to get nothing other than a zero Time to FIlament Change.

I have re-uploaded the files as mentioned elsewhere, and there still seems to be no change.

Here are pointers to some images of the panel and the settings and the gcode file before uploading for these (for some reason, github doesn't like my files and refuses to accept them)

https://www.dropbox.com/s/z1bi9y8s1p6hw8m/CR-10v2%20%28Blue%29%20%5BOctoPrint%5D%2010%25%2019%3A48%202020-05-27%2013-02-31.jpg

https://www.dropbox.com/s/vppn3o7ats02wqk/CR-10v2%20%28Blue%29%20%5BOctoPrint%5D%2010%25%2019%3A48%202020-05-27%2013-01-15.jpg

https://www.dropbox.com/s/0i37yx9w3j1lidg/Puzzle%20Chessboard%20Pieces%201%20%28x4%29%20%5BPLA%5D%20Adhesion-%5Bbrim%5D%20%7C%20Infill-%5B50%25%5D%20%7C%20NozDiam-%5B0.4%5D%20%7C%20LH-%5B0.12%5D%20%7C%20LW-%5B0.4%5D%20%7C%20ExtTemp-%5B215.0%C2%B0C%5D%20%7C%20BedTemp-%5B65.0%C2%B0C%5D_multi.gcode

I'm sure it's just an Eye-Dee-Ten-Tee error, but I'd appreciate telling this Eye-Dee-Ten-Tee where I'm being one.

KenLucke, 2020-05-28:
After it passes the filament change, the Zero time line disappears, which I ass/ume is normal
[image]

KenLucke, 2020-05-28:
Weird.  NOW, it's showing a time, but WAY off.  From experience, printing this file (I have to do it 16 times to get all 64 pieces for the chessboard), the filament change is 5:22:00 and change into the print.  Just restarted a new copy of the print, and it's telling me it's only 1:45:00 or so to go.

[image]

eyal0, 2020-05-29:
I need to look into it some more. Perhaps you can send me the gcode?

eyal0, 2020-05-29:
I put some debugging in.  Please follow these instructions: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/150#issuecomment-635727058

I hope that we can figure it out!

eyal0, 2020-06-02:
Maybe in fxing #3 I fixed your bug, too?  Can you install the latest version and try again?

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

Thanks!

KenLucke, 2020-06-12:
> Maybe in fxing #3 I fixed your bug, too? Can you install the latest version and try again?
> 
> https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip
> 
> Thanks!

Latest version still does not solve the issue.  Sorry it took so long to get back to you, had other issues with printers.

[image]
[image]

Relevant gcode section:
`G1 F2250 X147.083 Y198.55 E1431.98028
G1 F2700 E1429.98028
G1 F300 Z3.86
G0 F11250 X139.02 Y190.208 Z3.86
G1 F300 Z3.36
G1 F2700 E1431.98028
G1 F2250 X142.711 Y186.517 E1432.08445
G0 F11250 X141.576 Y187.087
G1 F2250 X139.581 Y189.081 E1432.14073
G1 F2700 E1430.14073
G1 F300 Z3.86
@Object NONMESH
G0 F11250 X144.387 Y112.696 Z3.86
;TIME_ELAPSED:3021.004989

;LAYER:28
M117 INDICATOR-Layer28
; multi color
M117 Change filament Now!
M600

@Object Full Set.3mf(10)
G0 X144.387 Y112.696 Z3.98
;TYPE:WALL-INNER
G1 F300 Z3.48
G1 F2700 E1432.14073
G1 F2250 X143.678 Y112.844 E1432.15519
G1 X143.077 Y113.057 E1432.16791
G1 X142.529 Y113.341 E1432.18023
G1 X142.035 Y113.675 E1432.19213
G1 X141.627 Y114.029 E1432.20291
G1 X141.151 Y114.564 E1432.2172
G1 X140.757 Y115.17 E1432.23163
G1 X140.521 Y115.674 E1432.24273
G1 X140.448 Y115.861 E1432.24674
G1 X140.356 Y116.139 E1432.25258`

KenLucke, 2020-06-12:
I **think** I may have found it...  Your sample code is "; layer" (with a space), whereas my code is ";layer" with no space. I will have to wait to start another print to confirm, but that sure looks as if it may be what is messing me up.

KenLucke, 2020-06-12:
> I **think** I may have found it... Your sample code is "; layer" (with a space), whereas my code is ";layer" with no space. I will have to wait to start another print to confirm, but that sure looks as if it may be what is messing me up.

Nope, still nothing.  Grrrr....

Are you looking for a particular CR/LF ending, perhaps?

KenLucke, 2020-06-13:
I figured out the problem with showing the time to next layer.  Your grep expression is case sensitive.  My files use "LAYER," your grep expression uses "layer."  Changing the lowercase to uppercase in your expressions (and removing the extraneous space) solved that portion of my issues.  You should be using "grep -i" (case insensitive) in your code.

I still cannot figure out why it is not showing the time to the next ^M600.

KenLucke, 2020-06-13:
FIXED IT!

I modified your search from
`^M600`
to
`^M600.*`
[added a ".*"]

and the example code to
`^M600.*`
[same addition]

WHY your simple search for 
`^M600`
matched your example of 
`M600 whatever`
and NOT my simple
`M600`
actual line, I have NO idea whatsoever.  I'll leave that for you to figure out.

KenLucke, 2020-06-13:
Now a wrinkle.  I removed the '*" [accidently], and it still works now, whereas it didn't with exactly the same parameters.  So I give up, it's working, if it ain't broke don't fix it

eyal0, 2020-06-13:
Weird. I'm just doing a normal python regular expression. `^M600` matches from the start of the line to then end of the M600. I'm not sure if characters after cause the match to fail or not. I don't recall how python works there.

Anyway, looks like you got it working! If you have a suggestion that would be universally useful, I can put it in the code so that everyone can enjoy it.

eyal0, 2020-06-13:
You can edit this file at the spot and then submit a PR. It's pretty easy.

https://github.com/eyal0/OctoPrint-TimeToFilament/blob/4a580f118284419b027991212401859b9908a23b/octoprint_TimeToFilament/__init__.py#L32

KenLucke, 2020-06-13:
Yeah, I know how it’s supposed to work :)

For some reason, it wasn’t.

eyal0, 2020-06-13:
:shrug: close the issue?

KenLucke, 2020-06-13:
As far as I am concerned, you can.

closed 2020-06-13 by eyal0