eyal0/OctoPrint-TimeToFilament #4: [Bug]

KenLucke, 2020-06-14:
Now that things are more or less working properly, I have found a slight bug.

Both the Time To Next Layer and Time To Filament Change are working accurately and both show up in the panel.  However, when the filament change line disappears, so does the time to next layer line, and it does not reappear unless I reload the octoprint tab.

I did have two tabs open for the same print (Dashboard in Full Browser mode and the gccode viewer) when I noticed this, although the Dashboard (in Full Browser mode) does not show the panel that contains the time estimates.  Don't know if the second tab open might or might not have anything to do with the issue, but I will do some checking the next time I have a filament change file running.

eyal0, 2020-06-14:
Ah yes. There is a bug here:

https://github.com/eyal0/OctoPrint-TimeToFilament/blob/4a580f118284419b027991212401859b9908a23b/octoprint_TimeToFilament/static/js/TimeToFilament.js#L128

eyal0, 2020-07-15:
Fixed by https://github.com/eyal0/OctoPrint-TimeToFilament/commit/6e4c9e2f05a20f4297d1563515d89249d3a4aade

commit 6e4c9e2f (2020-07-15): Only hide the regex that isn't found, not all of them.

This fixes #4
closed 2020-07-15 by eyal0