eyal0/OctoPrint-TimeToFilament #5: [Request] Add "Time OF Filament Change"

KenLucke, 2020-09-06:
Can you add an "Actual Time Filament Change" [or similar title] to the "Time to Filament Change" -  I get too lazy sometimes to do the math :)

```
Time to Filament Change: 00:02:34    <------- Time from now
Time of Filament Change: 23:16:08     <------- Actual Calculated time
```

Thank you for all your work!

eyal0, 2020-09-09:
This is doable without changing the code, even.  In the settings there is a part where it says something like this:

```
Filament change in <b>${formatDuration(this.progress.printTimeLeft - this.progress.TimeToFilament["^M600"].timeLeft)}</b>'}
```

That function in the middle needs changing.  Instead of `formatDuration`, it could be `time.strftime("%c")`

I don't have time to dig into it right now but if you want to try it, open up the settings and look for the Filament change line.  Replace:

`formatDuration(`

with:

`time.strftime("%c", time.gmtime(time.time()+`

Now you need to balance the parenthesis so replace the very last:

`)`

with:

`))`

I haven't tested it but that might work.  If you find that the output is too long, try replacing the `%c` with one or more of the percent codes listed here: https://docs.python.org/3/library/time.html#time.strftime.  You might also change the change the `Filament change in` to `Filament change at`.

You can also choose to not modify the existing one but instead copy and paste all the values into a new one that you create by clicking on the plus button.  And then you can have both available.  There is also a little checkmark that will allow you to disable and enable them as you go.

The setting dialog runs tests as you type so you should be able to see the results of your work while you're adjusting the settings.

Let me know how it goes and if you're still struggling with it then I'll dig into it some more when I have more time.  Good luck!

KenLucke, 2020-09-09:
OK, seeing how to do that (adding another display line, which is just perfect), but sorry, not completely sure the substitution.  I'm thinking the final `.timeleft`, as it appears to be relative, is going to be wrong.

this is what I did (but have not tried yet):
`Filament change at <b>${time.strftime("%c", time.gmtime(time.time()+["^M600"].timeLeft))}</b>'}`

Could you, when you get a chance, just post the string I need to use in entirety?

eyal0, 2020-09-09:
I think like this.  Again, untested:

```
Filament change at <b>${time.strftime("%c", time.gmtime(time.time()+this.progress.printTimeLeft - this.progress.TimeToFilament["^M600"].timeLeft))}</b>'}
```
I'll try to make time to test it.

KenLucke, 2020-09-09:
About to start a model with filament change, so I should know shortly.

KenLucke, 2020-09-09:
Nope.  No workie-workie :(

[image]

eyal0, 2020-09-10:
Oops, I wrote it in python but it should have been javascript. My bad.

```
Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft - this.progress.TimeToFilament["^M600"].timeLeft)*1000)}</b>'}
```

KenLucke, 2020-09-10:
OK, that one didn't throw an error when I entered it, but that print is already past the filament change, so I'll have to wait for the next one.  Thanks again for your work, and I'll let you know.

eyal0, 2020-09-10:
Well that's a start!  Did the output in the settings screen look reasonable?  I'm curious to see how it went.  Good luck!

KenLucke, 2020-09-10:
Close. I had to drop the final right brace and single quote (which I have done).  I also need to find the time format options (like the one you linked to for Python) for Javascript instead of Python, cause I don't need all of that info.   But it's on its way.

[image]

eyal0, 2020-09-10:
You could try formatDate, just like I used to use formatDuration.

https://github.com/OctoPrint/OctoPrint/blob/3ab84ed7e4c3aaaf71fe0f184b465f25d689f929/src/octoprint/static/js/app/helpers.js#L579

It's a function that is part of OctoPrint and those are accessible to you, just like I used formatDuration.  So I think that it's, like...

`formatDate(new Date(Date.now()` etc

and one more closing parenthesis at the end before `}</b>` etc

Something like that.  Time formatting is not so easy in javascript so people write their own functions.

Sekisback, 2021-01-17:
[image]

`Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)*1000).toLocaleTimeString('en-US', {hour12:false})}</b>`

eyal0, 2021-01-19:
I added it in 1.2.0.

Also, it should be `+` and not `-`.  Like this:

```
            {"enabled": False,
             "description": "Time of Next Filament Change",
             "regex": "^M600",
             "format": 'Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)*1000).toLocaleTimeString([], {hour12:false})}</b>'},
```

It's the current time, then add to that the total print time left.  That's when the whole print will be done.  Then subtract from that the print time left when we see M600.  That will be the timestamp of when we need to M600.  As you noticed, you must multiple by 1000 because the results are in seconds but the Date.now() is milliseconds.

If there's more for me to do here, let me know.

closed 2021-01-19 by eyal0

KenLucke, 2021-01-20:
> If there's more for me to do here, let me know.

Works beautifully, and got rid  of al the extraneous date/time zone/etc. crap, so it no longer line breaks.

[image]

Is there currently a way to get the number of [regex] remaining?  I.e., for "^M600," "3 Filament changes remaining," for "^M0," "7 Pauses remaining," etc

KenLucke, 2021-01-20:
Oh, and I just put tthem al in  the same format box for one:

```
Filament change in <b>${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)} </b>
<br>
Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)*1000).toLocaleTimeString([], {hour12:false})}</b>
```

eyal0, 2021-01-20:
Yup, that'll work just as well.

One advantage of separating them is that you can enable/disable them individually.