eyal0/OctoPrint-TimeToFilament #6: Please Release!

KenLucke, 2020-09-17:
This does not show up in the Plugin Manager "Get More" listing.  I believe you may have forgotten to actually release it.  Every time I have to install or do something with it, I have to come back here to get the install URL.

eyal0, 2020-09-18:
I tried to release it but they won't let me put it there. My code does some monkey patching of OctoPrint to provide this feature that would otherwise be impossible to provide. @foosel said that she might add a more proper way to do what I'm doing here but until then, I'm in limbo.

eyal0, 2020-09-18:
https://github.com/OctoPrint/plugins.octoprint.org/pull/553

KenLucke, 2020-09-18:
OK, Understood.

One other question:  Are you not using a standard REGEX implementation?

I tried this search string
`^M104`
and this format
`Temperature change in <b>${formatDuration(this.progress.printTimeLeft - this.progress.TimeToFilament["^M104"].timeLeft)} seconds</b>`

Which works, 

but using the search:
`^M104 S(\d+)`
and the format:
`Temperature change to \1 in <b>${formatDuration(this.progress.printTimeLeft - this.progress.TimeToFilament["^M104"].timeLeft)} seconds</b>`

(I'm sure you see what I'm trying to do there, pulling the actual temperature change and using it in the format.  This could also be used to specify at which layer a pause or filament change was going to occur.)

It seems that if there are any spaces after the M104 in the search criteria, it throws a fit, er, "exception"

Sample code used to test:
```
; This is a fake gcode file for testing your display lines.
; The first ^ in the file is the pretend current file position, otherwise just the first character.
; You can modify this.
G00 whatever
;LAYER:11
M12345 more pretend stuff
M600 ; whatever
M0 ; whatever
M104 S250 ; Change temperature to whatever
```

`

jneilliii, 2020-12-09:
Hey @eyal0 the latest release of OctoPrint now supports the additional data approach that we were discussing in your plugin PR. Have you taken a look yet and adapted your code to utilize that?

eyal0, 2020-12-09:
I haven't yet. I have to determine if it will actually do what I need and then adapt the code. It'll be a pretty big change but the code isn't so large so maybe it'll be alright. I'll start this week.

eyal0, 2020-12-09:
Here's what I will need to use: https://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-printer-additional-state-data

jneilliii, 2020-12-09:
Correct, and when submitting your PR you'll need to make sure to set the compatibility flag for OctoPrint to minimum 1.5.0.

eyal0, 2020-12-09:
I created a new version that uses the new API.  Thanks for the tip about OctoPrint minimum version.

When you install the new version, everything should work basically the same as it did before.

KenLucke, 2020-12-09:
> I created a new version that uses the new API. Thanks for the tip about OctoPrint minimum version.
> 
> When you install the new version, everything should work basically the same as it did before.

Link?

eyal0, 2020-12-09:
Install this:

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

I forgot one think: the settings will need adjusting.  Either press the button to reset to default settings or look for `this.progress.TimeToFilament` in the "format" strings in the advanced settings and replace each occurence with `this.plugins.TimeToFilament`.  Notice that `progress` becomes `plugins`, but only before `.TimeToFilament`.

KenLucke, 2020-12-09:
One thing I notice (and this is not new):  It fails to save any changes you make to the sample code.  For instance, I have to change 

`; layer 1`

to

`;LAYER:1`

every time I open the settings, because that's the way Cura formats the line layer numbers. (again, you should be case insensitive on searches if you can't use a standard regex library, which would eliminate at least the case matching failures :) ).  I also have to add the sample `M0` or other things for which I want it to report.

```
; This is a fake gcode file for testing your display lines.
; The first ^ in the file is the pretend current file position, otherwise just the first character.
; You can modify this.
G00 whatever
;LAYER:1
M12345 more pretend stuff
M600 ; whatever
M0 ; whatever
```

eyal0, 2020-12-10:
Yes, I know. I intentionally didn't save the sample text because I didn't think that it was important to save it. You think that I should try to save it? I expect that people won't be messing with it too much so isit worth while?

KenLucke, 2020-12-10:
Yes, I think it should be saved.  If you're tweaking things frequently, it's a pain to have to either edit, or copy/paste.

KenLucke, 2020-12-11:
I've had to disable my Time-To parameters.  I was getting interference with the gcode viewer, for some reason.  Without any of the 4 I have, the gcode viewer was smooth  and normal.  The more of the parameters I enabled, the jerkier it got, sometimes going 10 seconds or so without moving, then suddenly jumping to the new position (and leaving the normal corresponding lines behind).

https://www.dropbox.com/s/gn3piukdw9d6cc7/screencast%202020-12-11%2013-55-46.mp4

```
    TimeToFilament:
        displayLines:
        -   description: Time to Next Layer
            enabled: false
            format: Layer ${this.plugins.TimeToFilament["^;LAYER:(\\d+)"].groups[0]}
                start in <b>${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^;LAYER:(\\d+)"].timeLeft)}
                seconds</b>
            regex: ^;LAYER:(\d+)
        -   description: Time to Next Filament Change
            enabled: false
            format: Filament change in <b>${formatDuration(this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M600"].timeLeft)} seconds</b>
            regex: ^M600
        -   description: Time of Next Filament Change
            enabled: false
            format: Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M600"].timeLeft)*1000)}</b>'
            regex: ^M600
        -   description: Time to Next Pause
            enabled: false
            format: Printer Pause in <b>${formatDuration(this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M0"].timeLeft)} seconds</b>
            regex: ^M0
```

eyal0, 2020-12-11:
Interesting.  To be clear, it's the web GUI that's getting jerky and not the printer, right?  Maybe the code that I wrote in javascript isn't so efficient?  If it's just that, I think that I'd be able to debug it.  I will add a bunch of displayLines in mine, too, and see what happens.

KenLucke, 2020-12-12:
Yes, the printer is fine, the display is what's jittering.

eyal0, 2021-01-14:
It done.

https://github.com/OctoPrint/plugins.octoprint.org/pull/553

closed 2021-01-14 by eyal0