eyal0/OctoPrint-TimeToFilament #7: Large files cause problems

KenLucke, 2020-10-14:
Not sure where the size break point for the file is where this starts.  (See the end for what happens).

This is the file for the code below:
https://www.dropbox.com/s/n6vaffrzx461o8u/Control%20Box%20Storage%20Unit%20Final.gcode

```
Send: N28 G1 X50.0 Y5.3 Z0.28 F1500.0 E23*86
Recv:  T:225.19 /225.00 B:65.01 /65.00 @:33 B@:73
Send: N29 G92 E0*124
Recv:  T:224.81 /225.00 B:65.15 /65.00 @:42 B@:4
Send: N30 G1 Z2.0 F3000*40
Send: N31 G92 E0*117
Send: N32 G92 E0*118
Recv:  T:224.81 /225.00 B:65.26 /65.00 @:43 B@:0
Send: N33 G1 F3000 E-5*0
Send: N34 M107*18
Recv:  T:225.19 /225.00 B:65.20 /65.00 @:33 B@:0
Send: N35 G0 F11250 X79.584 Y39.89 Z0.288*37
Recv:  T:225.13 /225.00 B:65.08 /65.00 @:35 B@:0
Send: N36 G1 F3000 E0*45
Send: N37 G1 F1687.5 X79.994 Y39.093 E0.08585*41
Send: N38 G1 X80.078 Y38.944 E0.10224*89
Recv:  T:225.00 /225.00 B:65.00 /65.00 @:38 B@:22
Send: N39 G1 X80.913 Y37.635 E0.25096*87
Send: N40 G1 X81.201 Y37.232 E0.29841*93
Recv:  T:225.38 /225.00 B:64.86 /65.00 @:29 B@:89
Send: N41 G1 X81.769 Y36.492 E0.38777*80
Recv:  T:224.88 /225.00 B:64.79 /65.00 @:42 B@:127
Send: N42 G1 X81.886 Y36.35 E0.40539*99
Recv: //action:notification ETA is -
Recv: ok N1 P15 B7
Send: N43 G1 X82.48 Y35.685 E0.49081*106
Recv: ok N2 P15 B7
Recv: ok T:225.06 /225.00 B:64.77 /50.00 @:37 B@:127
Send: N44 G1 X82.602 Y35.557 E0.50774*92
Recv: ok T:225.06 /225.00 B:64.77 /50.00 @:37 B@:127
Send: N45 G1 X83.351 Y34.835 E0.6074*99
Recv: //action:notification CR10 V2 Ready.
Recv: //action:notification Bed Cooling...
Recv: //action:notification Printing...
Recv: ok N5 P15 B7
Send: N46 G1 X83.837 Y34.417 E0.6688*98
Recv: ok N6 P15 B7
Send: N47 G1 X84.243 Y34.086 E0.71898*94
Recv: ok T:225.13 /215.00 B:64.81 /50.00 @:35 B@:0
Send: N48 G1 X84.979 Y33.523 E0.80774*93
Recv: //action:notification Printing...
Recv: //action:notification Printing...
Recv: ok N8 P15 B7
Send: N49 G1 X85.109 Y33.43 E0.82305*97
Recv: ok N9 P15 B7
Recv: ok N10 P15 B7
Send: N50 G1 X86.005 Y32.849 E0.92534*81
Send: N51 G1 X86.924 Y32.344 E1.02579*93
Recv: ok N11 P15 B7
Send: N52 G1 X87.503 Y32.066 E1.08731*81
Recv: ok N12 P15 B7
Recv: ok N13 P15 B7
Send: N53 G1 X88.039 Y31.828 E1.14349*84
Recv: ok N14 P15 B7
Send: N54 G1 X88.875 Y31.488 E1.22994*90
Recv: ok N15 P15 B7
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Send: N55 G1 X89.052 Y31.423 E1.248*92
Recv:  0 +0.038 +0.011 +0.031 +0.063 +0.095
Recv:  1 -0.036 -0.049 -0.034 -0.022 +0.054
Recv:  2 -0.052 -0.065 -0.057 -0.049 +0.001
Recv:  3 -0.022 -0.078 -0.095 -0.077 -0.026
Recv:  4 +0.048 -0.043 -0.062 -0.018 +0.011
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 5.00
Recv: ok N16 P15 B7
Send: N56 G1 X89.896 Y31.149 E1.333*91
Recv:  T:224.88 /215.00 B:64.79 /50.00 @:0 B@:0
Send: N57 G1 X90.055 Y31.104 E1.34883*91
Send: N58 G1 X91.1 Y30.853 E1.45177*90
Send: N59 G1 X91.885 Y30.716 E1.52811*94
Recv:  T:225.25 /215.00 B:64.75 /50.00 @:0 B@:0
Send: N60 G1 X92.408 Y30.64 E1.57873*109
Send: N61 G1 X93.494 Y30.526 E1.68333*88
Recv:  T:224.44 /215.00 B:64.71 /50.00 @:0 B@:0
Send: N62 G1 X94.515 Y30.489 E1.78119*91
Send: N63 G1 X97.905 Y30.487 E2.10592*80
Recv:  T:223.69 /215.00 B:64.57 /50.00 @:0 B@:0
Send: N64 G1 X212.599 Y30.488 E13.09236*95
Send: N65 G1 X215.491 Y30.488 E13.36939*88
Recv:  T:222.63 /215.00 B:64.47 /50.00 @:0 B@:0
Send: N66 G1 X216.617 Y30.533 E13.47733*87
Send: N67 G1 X217.573 Y30.636 E13.56943*89
Recv:  T:221.06 /215.00 B:64.26 /50.00 @:0 B@:0
Send: N68 G1 X218.064 Y30.707 E13.61696*90
Recv:  T:219.79 /215.00 B:64.11 /50.00 @:0 B@:0
Send: N69 G1 X219.019 Y30.878 E13.70989*86
Send: N70 G1 X219.176 Y30.912 E13.72528*94
Send: N71 G1 X220.033 Y31.131 E13.81001*95
Recv:  T:218.07 /215.00 B:63.92 /50.00 @:0 B@:0
Send: N72 G1 X220.23 Y31.189 E13.82968*107
Recv:  T:216.77 /215.00 B:63.75 /50.00 @:10 B@:0
Send: N73 G1 X221.073 Y31.471 E13.91483*86
Send: N74 G1 X221.212 Y31.523 E13.92904*83
Send: N75 G1 X221.817 Y31.767 E13.99153*94
Recv:  T:215.31 /215.00 B:63.51 /50.00 @:27 B@:0
Send: N76 G1 X222.294 Y31.974 E14.04134*81
Recv:  T:213.44 /215.00 B:63.34 /50.00 @:53 B@:0
Send: N77 G1 X223.145 Y32.377 E14.13153*83
Send: N78 G1 X223.303 Y32.46 E14.14863*96
Send: N79 G1 X224.067 Y32.894 E14.2328*99
Recv:  T:212.08 /215.00 B:63.14 /50.00 @:72 B@:0
Send: N80 G1 X224.238 Y33 E14.25207*78
Send: N81 G1 X224.958 Y33.481 E14.33501*87
Recv:  T:211.82 /215.00 B:63.00 /50.00 @:69 B@:0
Send: N82 G1 X225.122 Y33.599 E14.35436*91
Recv:  T:211.67 /215.00 B:62.83 /50.00 @:68 B@:0
Send: N83 G1 X225.671 Y34.018 E14.42052*86
Send: N84 G1 X226.065 Y34.335 E14.46896*89
Recv:  T:211.98 /215.00 B:62.64 /50.00 @:62 B@:0
Send: N85 G1 X226.811 Y34.983 E14.56361*86
Recv:  T:212.55 /215.00 B:62.47 /50.00 @:55 B@:0
Send: N86 G1 X226.91 Y35.076 E14.57662*96
Send: N87 G1 X227.552 Y35.721 E14.6638*100
Send: N88 G1 X227.673 Y35.851 E14.68081*95
Recv:  T:214.11 /215.00 B:62.29 /50.00 @:26 B@:0
Send: N89 G1 X228.345 Y36.637 E14.77986*92
Recv:  T:214.95 /215.00 B:62.12 /50.00 @:16 B@:0
Send: N90 G1 X228.687 Y37.084 E14.83378*80
Send: N91 G1 X229.049 Y37.583 E14.89283*89
Send: N92 G1 X229.549 Y38.322 E14.9783*96
Recv:  T:215.68 /215.00 B:62.00 /50.00 @:8 B@:0
Send: N93 G1 X229.648 Y38.48 E14.99616*104
Send: N94 G1 X230.084 Y39.238 E15.07992*80
Recv:  T:216.04 /215.00 B:61.85 /50.00 @:7 B@:0
Send: N95 G1 X230.176 Y39.417 E15.0992*97
Recv:  T:215.99 /215.00 B:61.66 /50.00 @:12 B@:0
Recv:  T:215.52 /215.00 B:61.54 /50.00 @:23 B@:0
Recv:  T:214.69 /215.00 B:61.43 /50.00 @:41 B@:0
Recv:  T:213.75 /215.00 B:61.22 /50.00 @:58 B@:0
Recv:  T:213.80 /215.00 B:61.12 /50.00 @:53 B@:0
Recv:  T:213.85 /215.00 B:60.96 /50.00 @:51 B@:0
Recv:  T:214.17 /215.00 B:60.80 /50.00 @:44 B@:0
Recv:  T:214.64 /215.00 B:60.71 /50.00 @:37 B@:0
Recv:  T:215.21 /215.00 B:60.63 /50.00 @:27 B@:0
Recv: X:15.00 Y:30.00 Z:5.00 E:-4.00 Count X:1200 Y:2400 Z:2000
Recv: ok N17 P15 B0
Send: N96 G1 X230.818 Y40.82 E15.24699*103
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 5.00
Recv: ok N18 P15 B0
Send: N97 G1 X231.002 Y41.283 E15.29472*92
Recv: X:15.00 Y:30.00 Z:5.00 E:0.00 Count X:1200 Y:2400 Z:2000
Recv: ok N19 P15 B0
Send: N98 G1 X231.323 Y42.186 E15.38652*86
Recv: ok N20 P14 B0
Recv: ok T:215.31 /215.00 B:60.60 /50.00 @:28 B@:0
Send: N99 G1 X231.371 Y42.337 E15.40169*88
Send: N100 G1 X231.608 Y43.196 E15.48705*110
Recv: Error:checksum mismatch, Last Line: 27
Recv: Resend: 28
Printer requested line 28 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 28 but no sufficient history is available, can't resend"
Send: M112
Send: N101 M112*33
Send: N102 M104 T0 S0*34
Send: N103 M140 S0*103
Changing monitoring state from "Error: Printer requested line 28 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 28 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop
```

Here's the relevant lines from the above file (around lines 27), but the whole file is shown above.

```
1. ;FLAVOR:Marlin
2. ;TIME:311916
3. ;Filament used: 221.219m
4. ;Layer height: 0.2
5. ;MINX:77.699
6. ;MINY:30.286
7. ;MINZ:0.288
8. ;MAXX:232.3
9. ;MAXY:289.8
10. ;MAXZ:101.538
11. ;Generated with Cura_SteamEngine 4.6.2
12. M140 S50
13. M105
14. M190 S50
15. M104 S215
16. M105
17. M109 S215
18. M82 ;absolute extrusion mode
19. M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
20. M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
21. M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
22. M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
23. M220 S100 ;Reset Feedrate
24. M221 S100 ;Reset Flowrate
25. G28 ;Home
26. M420 S1 Z5 ; use stored mesh, bed level on, fade height 5mm
27. G92 E0 ;Reset Extruder
28. G1 Z2.0 F3000 ;Move Z Axis up
29. G1 X10.1 Y30.0 Z0.28 F5000.0 ;move away from clip
30. G1 X50.0 Y30.0 Z0.28 F5000.0 ;move around clip
31. G1 X50.0 Y5.0 Z0.28 F5000.0 ;Move to start position
32. G1 X250.0 Y5.0 Z0.28 F1500.0 E11 ;Draw the first line
33. G1 X250.0 Y5.3 Z0.28 F5000.0 ;Move to side a little
34. G1 X50.0 Y5.3 Z0.28 F1500.0 E23 ;Draw the second line
35. G92 E0 ;Reset Extruder
36. G1 Z2.0 F3000 ;Move Z Axis up
37. G92 E0
38. G92 E0
39. G1 F3000 E-5
40. ;LAYER_COUNT:609
41. ;LAYER:-6
42. M107
43. G0 F11250 X79.584 Y39.89 Z0.288
44. ;TYPE:SUPPORT-INTERFACE
```

KenLucke, 2020-10-14:
Forgot to say, it's definitely this plugin.  Disabling or removing it eliminates the issue, reinstalling or reenabling it causes it to return on the next large file.

eyal0, 2020-12-16:
I am also seeing a checksum error on N100.  I will investigate.

eyal0, 2020-12-16:
Oh, looks like the virtual printer has a default forced error on `N100`!  For testing, I guess.

https://github.com/OctoPrint/OctoPrint/blob/f67c15a9a47794a68be9aed4f2d5a12a87e70179/src/octoprint/plugins/virtual_printer/virtual.py#L426

So that's why I have an errors on 100, 105, 110, 115.  Funny!

Anyway, your log shows that it was trying to print N100 but N27 had a checksum error and by the time the printer got to N100, N27 was no longer in the buffer.  Bummer.  I wish that your log went just a little little bit more back so I could see that N27 but it looks like you started at N28.

Another thing worth checking is what if you just lop off the end of the file to see if the size is really the cause.

eyal0, 2020-12-16:
By the way, the line numbers in the gcode file aren't the same as the line number in the error.  The `N` prefix is the line number so what the printer called 27 is just the 27th line of gcode that was sent, not counting the lines that needn't be sent because they are just comments.

So gcode file line 33 is N27.  Looks like a plain old string, though, so I don't get how the checksum would be wrong in the first place!  That's weird.  And it looks like the OctoPrint buffer is too small.

If you had even more of the log, that would help.  And the test for file size that I mentioned above.