eyal0/OctoPrint-TimeToFilament #8: I've had to disable my Time-To parameters.  I was getting interference with the gcode viewer, for some reason.  Without any of the 4 I have, the gcode viewer was smooth  and normal.  The more of the parameters I enabled, the jerkier it got, sometimes going 10 seconds or so without moving, then suddenly jumping to the new position (and leaving the normal corresponding lines behind).

eyal0, 2020-12-12:
I've had to disable my Time-To parameters.  I was getting interference with the gcode viewer, for some reason.  Without any of the 4 I have, the gcode viewer was smooth  and normal.  The more of the parameters I enabled, the jerkier it got, sometimes going 10 seconds or so without moving, then suddenly jumping to the new position (and leaving the normal corresponding lines behind).

https://www.dropbox.com/s/gn3piukdw9d6cc7/screencast%202020-12-11%2013-55-46.mp4

```
    TimeToFilament:
        displayLines:
        -   description: Time to Next Layer
            enabled: false
            format: Layer ${this.plugins.TimeToFilament["^;LAYER:(\\d+)"].groups[0]}
                start in <b>${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^;LAYER:(\\d+)"].timeLeft)}
                seconds</b>
            regex: ^;LAYER:(\d+)
        -   description: Time to Next Filament Change
            enabled: false
            format: Filament change in <b>${formatDuration(this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M600"].timeLeft)} seconds</b>
            regex: ^M600
        -   description: Time of Next Filament Change
            enabled: false
            format: Filament change at <b>${new Date(Date.now() + (this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M600"].timeLeft)*1000)}</b>'
            regex: ^M600
        -   description: Time to Next Pause
            enabled: false
            format: Printer Pause in <b>${formatDuration(this.progress.printTimeLeft
                - this.plugins.TimeToFilament["^M0"].timeLeft)} seconds</b>
            regex: ^M0
```

_Originally posted by @KenLucke in https://github.com/eyal0/OctoPrint-TimeToFilament/issues/6#issuecomment-743449208_

eyal0, 2020-12-12:
Sorry about that.  It was a regression.  While working on #6, I re-broke something.  Should be fine now.  Try version 1.1.1 and let me know, thanks!

I'm still trying to get approved for the plugin repo.

closed 2020-12-12 by eyal0

KenLucke, 2020-12-12:
will do, once current prints are done.

PR #9 (2020-12-12): Remove Infinities from sent result but not from cache
Removing them from the cache causes the file to be searched again each time, which is very slow.

This is a regression caused by c7425eeb25bb5b287439084fdb6001e7131282aa

This fixes #8
commit 06a55cfc (2020-12-12): Remove Infinities from sent result but not from cache

Removing them from the cache causes the file to be searched again each time, which is very slow.

This is a regression caused by c7425eeb25bb5b287439084fdb6001e7131282aa

This fixes #8