eyal0/OctoPrint-TimeToFilament #11: Filamentchnage in 00:00:00

Sekisback, 2021-01-15:
hey there, in my case the display shows only 00:00:00

Gcode relevant part
`
;TIME_ELAPSED:228.722401
M600 E30.00 L300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:4
M117 INDICATOR-Layer4
;TYPE:WALL-INNER
...

`

plugin Settings default

[image]

Sekisback, 2021-01-15:
after a restart it works like expected

eyal0, 2021-01-15:
Great!  You're one of the first people to try this out so if you have and suggestions or problems, let me know.

Sekisback, 2021-01-16:
Hey eyal0, I was happy to early. So I'll try to explane what i tested out so far. After playing around with the setting, i deinstalled and reinstalled the plugin, after that i had the same issue above. 
Normaly I use Ultimaker Format Package (.ufp) direct from Cura. I sliced a new on as GCode in both cases i am not able to get the time working. Plugin Settings default. 
Raspberry Pi 3 Model B Rev 1.2
octopi_version: 0.17.0
python.version: 3.7.3
octoprint.version: 1.5.2
printer.firmware: Marlin 2.0.7.2

[image]

[image]
[image]

This was yesterday on a linux system with default language: english
Today i switched to Windows 10 OS System language: german
and here there is a strange plugin behavior 

[image]
[image]

I guess this is a translation issue. If you found it out and if you need a hand for the german translation, let me know

eyal0, 2021-01-16:
Looks like it is detecting that you have M600 in your gcode but when it is computing the time until the M600, it says 0.

I think that the next step would be for me to add some debugging so that you will have some useful information in the logs. And then you can send me the logs.

We will try that next.

Sekisback, 2021-01-16:
the confusing part is, that i saw it working once yesterday. I tried it with this cura 3.8 generated gcode...
[cura.zip](https://github.com/eyal0/OctoPrint-TimeToFilament/files/5824681/cura.zip)

eyal0, 2021-01-16:
Seems like maybe two different issues:

1. Filament change time is 0.
2. When you switch languages, something isn't working right.

Let's fix the first one.  I did testing on your file and it works fine at the beginning until the filament change arrives.  I will show you:

[image]

It says filament change time is 0.  Also, below, there is a button that says "Resume".  The printer is waiting for me to push resume because we have reach the filament change time.  I am supposed to change the filament and then push resume.  After I push resume, then it continues.  There are no more filament changes after that so it does not say filament change again after.

Can you please enable logging?  In the settings, go to the logging screen and add a row for debug logging of TimeToFilament:

[image]

You see the row that I added for TimeToFilament.  You should do the same.  Also, make sure that you press the blue plus sign or else the change will not take effect.

Now it will create a log file in `~/.octoprint/logs/plugin_TimeToFilament_engine.log`.  Can you examine that one?  And send it if you can, too.  Here's what mine looks like (you will have different numbers but the `M600` will be the same:

```
2021-01-16 11:03:46,843 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:03:57,089 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:07,282 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:17,329 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:27,374 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:37,420 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:47,467 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:04:57,514 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:05:07,561 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:05:17,609 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
2021-01-16 11:05:27,656 sending: {"^M600": {"timeLeft": 1202.2569749059626, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 24296, "searchPos": 258}}
```

Thanks!

Sekisback, 2021-01-16:
sure attached the log file

[plugin_TimeToFilament_engine.log](https://github.com/eyal0/OctoPrint-TimeToFilament/files/5824880/plugin_TimeToFilament_engine.log)

eyal0, 2021-01-16:
Except for the error on the first line, it looks fine.  It seems to have recovered.

When I try to print the file that you made, first I see that the filament change time is mor than 0 and then it decrease until it reaches zero.  And then it waits for me to push "Resume".

When you print, is the filament change time 0 even at the very beginning?

Sekisback, 2021-01-17:
Morning, yes you are right, the filament change time is zero right from start
[image]

eyal0, 2021-01-18:
I fixed the language issue, it was a bug in my code, and it seems like yours is working now, too.

If you need more help, open another issue, thanks.

closed 2021-01-18 by eyal0

JuPaChKo, 2021-01-31:
Ich sende den Gcode von Cura direkt an einen Oktoprint, da die Zeit bis zum Wechsel des Filaments. Drucke ich das Teil Teil bleibt die Zeit bei 0 stehen. Muss ich noch etwas ändern oder ist es ein Fehler.