eyal0/OctoPrint-TimeToFilament #13: No entries in Settings, no time Display

Master-68, 2021-01-18:
Doesn't seem to work... Running on OctoPI (all current updates for current stabe version 1.5.2 installed, Current version of PrintTimeGenius is also installed)

Pls. see screenshots:

[image]
[image]
[image]

System information:

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 4041732096
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Prusa-Firmware 3.9.2 based on Marlin

eyal0, 2021-01-18:
Seems like the same problem as #11. Does it work in English? He also had German.

I'm not sure how to configure German. How do I confugure German so that I can test this also?

Master-68, 2021-01-18:
To be honest, I'm not really sure how to change the language... I guess it should be possible to change the language here, but I tried to change to english and it had no effect. But maybe I just haven't installed the correct language packs when I set up OctoPI...

[image]

Master-68, 2021-01-18:
Just found, that you can change the language in the OctoPrint User settings:

[image]

It seems the settings are displayed, when language is set to English!

commit e300984a (2021-01-18): Search for last element instead of using the text

This fixes #13
This is related to #11
closed 2021-01-18 by eyal0