eyal0/OctoPrint-TimeToFilament #14: Error during installation

tchellomello, 2021-01-19:
I'm hitting the error below when using the `Plugin Manager` to install this extension:

[image]

```python
[octoprint-7b664bd9c-jdrl5] 2021-01-19 11:30:04,237 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip 
[octoprint-7b664bd9c-jdrl5] 2021-01-19 11:30:04,280 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading. 
[octoprint-7b664bd9c-jdrl5] 2021-01-19 11:30:05,481 - octoprint.plugins.pluginmanager - ERROR - Installing the plugin from https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip failed, could not parse output from pip. See plugin_pluginmanager_console.log for generated output 
```
```
root@octoprint-7b664bd9c-jdrl5:/octoprint/octoprint/logs# cat plugin_pluginmanager_console.log
2021-01-19 11:28:32,023   /usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp55aao3nu/OctoPrint-TimeToFilament-master.zip --no-cache-dir
2021-01-19 11:28:33,157 > Processing /tmp/tmp55aao3nu/OctoPrint-TimeToFilament-master.zip
2021-01-19 11:28:33,761 ! ERROR: Could not find a version that satisfies the requirement OctoPrint-PrintTimeGenius (from octoprint-timetofilament)
2021-01-19 11:28:33,762 ! ERROR: No matching distribution found for OctoPrint-PrintTimeGenius
2021-01-19 11:30:04,468   /usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmphfutzqqh/OctoPrint-TimeToFilament-master.zip --no-cache-dir
2021-01-19 11:30:05,480 > Processing /tmp/tmphfutzqqh/OctoPrint-TimeToFilament-master.zip
2021-01-19 11:30:05,481 ! ERROR: Could not find a version that satisfies the requirement OctoPrint-PrintTimeGenius (from octoprint-timetofilament)
2021-01-19 11:30:05,481 ! ERROR: No matching distribution found for OctoPrint-PrintTimeGenius
```

tchellomello, 2021-01-19:
The same happens whenever creating a new python virtual environment and running the install process.
```python
(py39) mdemello@storm ~> python -m venv /tmp/test                                                                                                                                                               11:35:28
(py39) mdemello@storm ~> source /tmp/test/bin/activate.fish                                                                                                                                                     11:35:40
(test) mdemello@storm ~> python --version                                                                                                                                                                       11:35:49
Python 3.9.1
(test) mdemello@storm ~> which python                                                                                                                                                                           11:35:55
/tmp/test/bin/python
(test) mdemello@storm ~> pip install octoprint                                                                                                                                                                  11:36:12
Collecting octoprint
  Using cached OctoPrint-1.5.3-py2.py3-none-any.whl (3.8 MB)
...SNIP....
(test) mdemello@storm ~> pip install "https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip"                                                                                                     11:36:42
Collecting https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip
  Using cached https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip
ERROR: Could not find a version that satisfies the requirement OctoPrint-PrintTimeGenius (from OctoPrint-TimeToFilament==1.2.0) (from versions: none)
ERROR: No matching distribution found for OctoPrint-PrintTimeGenius (from OctoPrint-TimeToFilament==1.2.0)
```

tchellomello, 2021-01-19:
The error is because the PrintTimeGenius is not installed, therefore it fails. How about adding this on the requirements.txt so it can be automatic for users?

eyal0, 2021-01-19:
> The error is because the PrintTimeGenius is not installed, therefore it fails. How about adding this on the requirements.txt so it can be automatic for users?

It already is I think.  Here are the contents of requirements.txt:

```
###
# This file is only here to make sure that something like
#
#    pip install -e .
#
# works as expected. Requirements can be found in setup.py.
###

.
```

The comment implies that I should put it in `setup.py`.  I did:

```
# Any additional requirements besides OctoPrint should be listed here
plugin_requires = ["OctoPrint-PrintTimeGenius", "OctoPrint>=1.5.0"]
```

So it *was* automatic, your install process tried to find OctoPrint-PrintTimeGenius.  But it failed to find it because it didn't know where it get it.  Your install of pip probably only knows to search at pypi.org, which is the default.

The only remedy to this would be to have pip know how to search the octoprint plugin repo.  I'll look into this and maybe suggest a solution to @foosel

foosel, 2021-01-19:
You could just publish PTG to pypi. It's fairly easy, you could even automate it through github actions.

tchellomello, 2021-01-19:
@eyal0  something like this should work
```
requirements.txt
git+git://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip
```

eyal0, 2021-01-19:
Is it good to have a link to master.zip?  I worry that it would break all versioning.  Like, what if I made an update to TTL that requires a newer version of PTG?  I think that it would be a problem, right?

I guess that I could publish to pypi, it's just more work for me!  Oh well, I'll look into it.

Is there a way to have requirement that also knows about versions and the github releases page?

foosel, 2021-01-19:
> Is there a way to have requirement that also knows about versions and the github releases page?

No, not unless I somehow find a way to turn the plugin repository itself into a valid python package index, and frankly I have a bit too much other stuff already on my plate to tackle this short or mid term.

tchellomello, 2021-01-19:
> Is it good to have a link to master.zip? I worry that it would break all versioning. Like, what if I made an update to TTL that requires a newer version of PTG? I think that it would be a problem, [...]

You can pin for a particular release, branch or commit hash as well. 

Check this thread for some examples: https://stackoverflow.com/questions/16584552/how-to-state-in-requirements-txt-a-direct-github-source#35998253

eyal0, 2021-01-19:
I don't want to pin, though.  I want to be able to write something like `OctoPrint-PrintTimeGenius>=1.2.3`.  And it should then grab the correct release.

Is that possible?  I guess that I could write some code into my `setup.py` to search through the github releases page and find an acceptable one.  And then add the URL of that acceptable one to the call to `setuptools.setup()`?

How does that sound?

foosel, 2021-01-19:
You are replicating functionality that is readily available in pip. And you are relying on `setup.py` actually being executed (read: your plugin being installed from a source distribution, which it doesn't necessarily have to). Don't do this to yourself when an alternative is *this* easily available. 

An automatic push of your plugin to pypi on every single release of your plugin is as simple as this:

``` yaml
name: Publish
on:
  release:
    types: [published, prereleased]

jobs:
  publish:
    name: Publish plugin
    runs-on: ubuntu-latest
    steps:
      - uses: actions/checkout@v2
        with:
          fetch-depth: 0
      - name: 🏗 Set up Python 3.7
        uses: actions/setup-python@v1
        with:
          python-version: 3.7
      - name: 🏗 Install build dependencies
        run: |
          python -m pip install wheel --user
      - name: 🔨 Build a binary wheel and a source tarball
        run: |
          python setup.py sdist bdist_wheel
      - name: 📦 Publish to index
        uses: pypa/gh-action-pypi-publish@master
        with:
          password: ${{ secrets.pypi_password }}
```

If you want to add automatic testing and so on in between, take a look at [OctoPrint's own build file](https://github.com/OctoPrint/OctoPrint/blob/master/.github/workflows/build.yml).

This even works for RCs. The only thing you have to make sure is that you version your stuff properly (with PEP440 compatible version numbers) and that you publish a new release if there's something wrong with the first attempt (pypi does not allow to revise releases).

eyal0, 2021-01-19:
I lost access to my Pypi account because I don't have my security key anymore.

I'll do this solution if I one day need versioning on PTG for TTL.  In the meanwhile, I'll just do a solution similar to what @tchellomello suggested.

commit eff54eb0 (2021-01-19): Require PrintTimeGenius with a link to the download

This fixes #14
closed 2021-01-19 by eyal0