eyal0/OctoPrint-TimeToFilament #15: Not working with PrintTimeGenius?

bordermultimedia, 2021-01-20:
Hi, on the settings pannel I see it works, but on the main screen it always shows "Filament change in 00:00:00", I have PrintTimeGenius enabled, could this be the cause of the bug? I'm sending the gcode in a zip file just in case (It was sliced with Cura for an Ender 3)

Thanks for the plugin!
[CE3_fichas-BNRtodas.zip](https://github.com/eyal0/OctoPrint-TimeToFilament/files/5839377/CE3_fichas-BNRtodas.zip)

eyal0, 2021-01-20:
PrintTimeGenius is actually a requirement for this plugin so you're okay on that front.

You're not the first to report that there is a problem where it shows always zero.  But I haven't been able to see the problem myself.  Are you on the latest version?  I made an update just a few hours ago.  Please check.  1.2.1 is the latest.

The other suggestions that I have is that you enable debug logging in the OctoPrint settings and send me some of the long.  Also, if you press F12 in your browser, you can see if there are any errors in the output.

Try whichever of those you are able and let me know if something helps.  Thanks!

waterfox207, 2021-01-21:
Hi,
I have the same Problem the displayed time to the next filament change is also the printing start time.
[image]

eyal0, 2021-01-22:
Would you please turn on debug logging and send me the contents of the log?  You can find debug logging in the settings.  Don't forget to click on the blue plus symbol before you save.

bordermultimedia, 2021-01-22:
I just checked that I had it activated, I'm sending you the log of that day
I printed several multicolor gcode files.

Hope this helps! Thanks!

eyal0, 2021-01-22:
I hope it helps, too.

In the past, I've had problems with foreign languages.  If you can, please try with the English language, too?  You can change the language in the OctoPrint settings.  Let me know if that makes a difference.  Thanks!

bordermultimedia, 2021-01-22:
I already use it in English though it was configured to auto-detect. I just
changed it to English only, and tried again with the virtual printer, same
results ( Filament change in *00:00:00*  ). Also, if I enable Time of next
filament, it changes constantly and is showing the current time minus 1.3
minutes or so:

[image: image.png]
I've tried it in the two instances of octoprint that I have running on my
Raspberry Pi 3 Model B Rev 1.2, and it shows the same.

Anything else I can try, just let me know. Thanks!

eyal0, 2021-01-23:
I don't see the log nor the image.  Can you try again?  I'm not sure if send from email will work for attachments.

bordermultimedia, 2021-01-23:
Sorry, let's try again, this was the image:
[image]
[plugin_TimeToFilament_engine.log.2021-01-19.zip](https://github.com/eyal0/OctoPrint-TimeToFilament/files/5859551/plugin_TimeToFilament_engine.log.2021-01-19.zip)
And here's the log file ;)

eyal0, 2021-01-25:
Hmm.  I do see valid lines in the log but I can't tell why they aren't doing anything useful in your OctoPrint.  The log has lines like this:

```
2021-01-20 03:06:43,988 sending: {"^M600": {"timeLeft": 28002.74196011205, "searchPos": 0, "groups": [], "groupdict": {}, "group": "M600", "matchPos": 11173604}}
```

A line like that means that M600 was found on a line and the M600 is at 28002 seconds before the end of the print.  That's about 7.75 hours!  But I see that your print is much shorter than that so I don't understand why it would say such a big number.  This is causing a negative number and that is why it says 0.  :-(

I think that maybe the problem is in PrintTimeGenius, though I don't know how it happened.  Want to try to re-analyze the file in PrintTimeGenius?  And send me the log of that, please.  If that doesn't help then I'll add more debug logging to this plugin and we'll try that.

This is a tough one!

waterfox207, 2021-01-25:
I have put @bordermultimedia 's .gcode file into my octoprint instance and logged the PrintTimeGenius process. 
Here is the log:
[octoprint (2).log](https://github.com/eyal0/OctoPrint-TimeToFilament/files/5866802/octoprint.2.log)

Hope that will help you!