eyal0/OctoPrint-TimeToFilament #16: Is there currently a way to get the number of [regex] remaining?

eyal0, 2021-01-20:
> If there's more for me to do here, let me know.

Works beautifully, and got rid  of al the extraneous date/time zone/etc. crap, so it no longer line breaks.

[image]

Is there currently a way to get the number of [regex] remaining?  I.e., for "^M600," "3 Filament changes remaining," for "^M0," "7 Pauses remaining," etc

_Originally posted by @KenLucke in https://github.com/eyal0/OctoPrint-TimeToFilament/issues/5#issuecomment-763270211_

eyal0, 2021-01-20:
Not currently.  The way it work is I look at the current file position and search forward just until I find the next match of the regex.  I then save the start position and the match position so that future searches can be sped up.

But it quits as soon as it finds one, yes.  I might be able to have it search all the way forward... Hmm.  I'll think about it.  I don't want to be doing that search too often because it'll slow down the computer.  But I could do it once and then remember the answer.  If the file is really large then it might potentially slow down the system doing the very first scan.

I'll think on it.

KenLucke, 2021-04-03:
How's the thought process on this?  :)

puterboy, 2022-08-26:
What about storing the number of matches to date? This shouldn't add any search time and could be useful for example if you want to display the next pause number or next color change number or even the next layer number.

eyal0, 2022-08-30:
Searching the whole file upfront is *slow*.  I worry that it would lock up the OctoPrint system for a while at the start of every print and that would be annoying!  Then again, if we do it while the printer is heating up, maybe no big deal?  We need to be careful about locking up OctoPrint for too long because the printer might think that OctoPrint has disconnected if OctoPrint doesn't communicate with the printer for too long.

puterboy, 2022-08-30:
Even if you don't search forward, per my previous comment, could one at least count the matches to date and store that in an accessible variable. That way the display message could say "Time for Nth color change"

eyal0, 2022-08-30:
It might not be accurate, though.  I always start the search from the current file position and the search runs regularly.  So it's possible that the current file position advanced faster than the runs of the function (once per 1-2 seconds, I think) and we skip over some lines?

We ask the printer for the current file position: https://github.com/eyal0/OctoPrint-TimeToFilament/blob/master/octoprint_TimeToFilament/__init__.py#L103 (It moves forward but it can skip over lines because it isn't running once per line, it's like once per 1-2 seconds.)

We start searching from there: https://github.com/eyal0/OctoPrint-TimeToFilament/blob/master/octoprint_TimeToFilament/__init__.py#L113

puterboy, 2022-08-30:
I meant to say count the matches that have already occurred (not found) -- i.e., if you are at position 20000, and three pauses have already occurred, store the number 3 for that regex. Then after the position passes the next occurrence of the regex, increment the number to 4. i.e., you are counting matches that have already been passed by the current position.

eyal0, 2022-08-30:
I understand.  I'm just saying that the count might be wrong if the thing that you are counting is in the file more than once per 1-2 seconds.