eyal0/OctoPrint-TimeToFilament #19: Potential conflict with DisplayLayerProgress

PRGeno, 2021-02-16:
I seem to have a difficult time getting the TimeToFilament (TTF) items to show up in the Status display with the DisplayLayerProgress (DLP) plugin enabled.  Disabling the DLP "Status" component first, then enabling TTF, then refreshing allows the TTF lines to sometimes show up (usually, but not always).  Then again sometimes after that I can re-enable the DLP "Status" component and both DLP and TTF items show, but not always.

There seems to me to be some sort of conflict there. DLP has been in all my OctoPrint systems for a long time and have become very dependent upon it.  TTF looks like it could also be quite useful but DLP has to work for it to be so.

Here is the Github link to DLP if you're not already familiar with it:

[](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress)

eyal0, 2021-02-17:
I've been trying to track down a bug that causes this plugin to fail for some users.  It never fails for me but maybe the problem is an interaction with other plugins.

Do you find that TimeToFilament works correctly when DLP is off?

eyal0, 2021-02-17:
I just tried using DLP together with TTF.  Worked fine.  :shrug: I'm not sure what to do.

PRGeno, 2021-02-17:
I'm not sure what to do either.  

I have two identical OctoPrint installations running on identical HP ProDesk Mini PCs running Ubuntu 19.x.  The have exactly the same plugins, all configured exactly the same connected to identical Prusa Mk3s Bear upgrade printers.

I have gone to great pains to make these identical to the point where the second one was created from a disk image created from the first one once I got it setup just the way I wanted.  The only thing different between them is the IP address and the color theme used in Themify, one is a black theme and one is red, to match the printers and to be easily identified in two tabs in my browser (Chrome).  

They have been running well for quite awhile and are not new installations, but whatever I do to one, I always do to the other, to keep them identical and DsiplayLayerProgress has been installed on both forever as has been PrintTimeGenuis.  TimeToFilament is the new edition to both. 

I first started trying to get BlackBear running TimeToFilament first since RedBear was running a job.  Nothing would show up from TimeToFilament in the status pane at first.  I made sure PrintTimeGenius was doing it's thing and it seemed to be.  The I started trying to find a conflict.  I started with disabling plugins the see if there was a conflict.  Then tried disabling the status part of DisplayLayerProgress (did not disable DisplayLayerProgress entirely).  I noticed TimeToFilament fields showed up after refreshing the browser tab.  I then enabled the status portion of   DisplayLayerProgress again and it all worked, just like I wanted.

I thought I figured out the magic sequence with TimeToFilament needing to get a hold of the status pane first and letting DisplayLayerProgress tag on after.

But then I tried it on RedBear and no fiddling around with the sequence or enabling or disabling thing will get TimeToFilment to show up in the status pane.  So now I don't know what is going on.  BlackBear is still fine, but RedBear never displayed the entries.

I have not dug into what is going on behind the scenes yet, and I'm at work now so I don't have access to the systems.  But I'll try to give you whatever you need if I can to try to help you track down what is going on.