eyal0/OctoPrint-TimeToFilament #22: [Feature Request] - Sortable Parameter list

KenLucke, 2021-04-04:
Currently, I can't seem to reorder my parameters list other than to delete and re-enter them in the order I want, or by editing the config.yaml file to do so.  It would be nice if I could reorder the list (and therefore the display order) directly in the settings.

Also, it would be nice to have a comprehensive list of the various Time() functions and their parameters to refer to when trying to solve a display format.  For instance, I'd like this:
`End of print: Tue Apr 06 2021 11:00:02 GMT-0700 (Pacific Daylight Time)]
`

to display without the Time Zone data, so it looks like this:
`End of print: Tue Apr 06 2021 11:00:02
`

Here's my current parameters, adapted from others.    I'm not familiar enough with the Date/Time functions to know what paraters to add/remove from them.  A list on the home page would be most helpful.
`End of print: <b>${new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^;End of Gcode"].timeLeft)*1000)}</b>]`

eyal0, 2021-04-06:
It's just javascript so if you want to trim off the end of a string, you can write javascript code.

```
End of print: <b>${String(new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^;End of Gcode"].timeLeft)*1000)).substring(0,24)}</b>]
```
So just the first 24 characters, maybe that'll work?

KenLucke, 2021-04-06:
Yes, that works, thanks.  I don't do javascript.  My actual coding days are long past, and I have not kept up on the languages, and I doubt the average user has it under their belt, either :)  - that's why I thought a table of various date/time variables would be helpful.