eyal0/OctoPrint-TimeToFilament #23: Using this with Pause at Height

anonymoususer321, 2021-04-17:
Hello. Thanks so much for making this plugin. I am trying to use it to tell when the next pause at height command will occur on my printer.  I am using the Cura "Pause at Height" script.  However, I cannot figure out where I need to configure things in your plugin to have a countdown until the next pause at height.  In case it helps, I have the following parameters in the script:

Method: BQ (M25)
G-code Before Pause: M300
G-code After Pause: M300

Thanks for your assistance with this.

eyal0, 2021-04-17:
Start by taking an existing item in the settings and copy it into a new one.  For example, there is one for `M600`.  I would take that one and make a copy of it and then replace all the `M600` with `M300`.  Make sure to replace *all* of them.  You just need to swap the 6 for a 3, I think.

Give that a shot, see if it works.

Bababoosky1, 2021-04-17:
Thanks so much for the prompt reply. So I clicked the "+" and then replicated one of the ones that were already there for M600.  Here is what I have.

***********************************************************
Description: Time of Next Pause at Height
Regex: ^M300
Format: Next Pause in <b>${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^M300"].timeLeft)}</b>
Sample Output: '^M300' not found in this.plugins.TimeToFilament
***********************************************************

Given that it says M300 is not found, is there somewhere else that I need to list M300 beyond what I have done above?

Thanks

Bababoosky1, 2021-04-17:
Actually, nevermind. Even though it says "not found", it is displaying properly in the "State" information panel!

eyal0, 2021-04-17:
The not found error is for the testing text at the very bottom.

In order to help test new settings, you can copy and paste some fake gcode into the cox at the very bottom of the settings. Your fake gcode didn't have M300 in it but you can add that.

Let me know if you want help modifying the display. Also, it's written in a standard regex and javascript language so anyone that knows regex or javascript could help you with it.

eyal0, 2021-04-17:
Reopen if you want more help.

closed 2021-04-17 by eyal0

Bababoosky1, 2021-04-17:
Sorry. I think I spoke too soon.  It is displaying the assigned name in the "State" information panel (i.e., "Next Pause In: 00:00:00", but it always says 00:00:00 (not the minutes remaining).  Is there something else that I need to configure to make it show the numbers counting down until the next pause?

eyal0, 2021-04-17:
This is a bug that I'm actively trying to fix. I need help from the community to debug. If you want to join in, read #21 and read it and follow the instructions at the bottom.

If you are able to help out, let's continue the conversation there so that we can keep these issue sorted. Thanks in advance for your help!