eyal0/OctoPrint-TimeToFilament #24: Not an issue :)

KenLucke, 2021-04-18:
Not an issue, just showing a bit of color I added:

[image]

eyal0, 2021-04-19:
Nice one.  Did you modify the settings so that there would be `<span color=red>` tags in the output?  Or something like that?

KenLucke, 2021-04-19:
Yep, exactly what I did.

```
Filament change in <b><span style="color:lime">${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)} </span></b>
<br>
Filament change at <b><span style="color:orange">${new Date(Date.now() + (this.progress.printTimeLeft - this.plugins.TimeToFilament["^M600"].timeLeft)*1000).toLocaleTimeString([], {hour12:false})}</span></b>
```

closed 2021-04-19 by eyal0