eyal0/OctoPrint-TimeToFilament #26: Filament Chage in shows "00:00:00"

jk257, 2021-08-10:
Hello,

Thank you in advance for your hard work on this. I've just started getting into multi-color printing and this is a huge tool for staying on top of my prints. I'm having an issue where it doesn't seem to be pulling the data correctly so the "change in" time is zero and therefore the "change at" time is also off. Attached is a screen shot of my settings and the dash panel. 
[image]

SenseGaming, 2021-08-12:
I have the same Issue.

Wyzed17, 2021-08-16:
Same thing over here. 

I've seen the log files of both print time genius and TimeToFilament. Print time Genius seems to do nothing wrong despite of this line:

`2021-08-16 08:42:13,851 Sarge output: Linear advance enabled
M900 K value ignored: Out of range (0-10)
...Linear advance disabled`

But I think that this has nothing to do with the final result, since I get a successful and accurate print time estimation.

Then we have the log of TimeToFilament, which shows this:

`2021-08-16 11:04:03,213 sending: {"^M600": {"timeLeft": 17179.974345518687, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 246182, "searchPos": 0}}`

The same exact line every ten seconds. "TimeLeft", in seconds, is larger than the time than Print Time Genius returns, and thus causes a bug, showing the reaming time 0:00:00.

`Processed 64467 Gcodes and 1231 Mcodes. 62937 blocks
Total time: 16672.219126`

troypouliot, 2021-09-11:
Same thing here. It never shows a time for the 1st filament change on the first print attempt. However, if I reprint the same file a 2nd time then the time to the 1st filament change will typically show up. weird. 
If I have a print that has multiple changes then the time to the 2nd and all after that show up correctly.

hartmannf, 2021-09-17:
got the same issue on prusa i3 MK3s & MMU2s
I noticed that in the generated Gcode there is no M600 comand. Changed Regx 2 M900 same problem not showing time 2 fil change. 00:00:00
Somebody an idee?

rayge99, 2021-12-02:
I'm seeing the same issue. 
I'm looking at a time left of 4 hours
Filament change in 00:00:00
Filament change at 09:50:21

So It is miss representing the time.

maybe there is a plugin conflict between it and something else we are not aware of yet?

nonewillescape, 2021-12-13:
Hi! Many thanks for this unique plugin for the folkx without MMU.
I'm also currently seeing the same 00:00:00 issue and inacurate filament change at time.
If it can be of any help, the plugin was working flawlessly on my MK3 before I decided to upgrade & update 3 days ago.

Here are the changes I have made to my setup:
- Upgrading the MK3 to a MK3S+
- Updating PrusaSlicer from 2.3.1 to 2.3.3
- Updating slicer profiles from Pretty PETG V3 to V4
- Updating my Printer Profiles in Octoprint to reflect the change (mostly the name and the Print bed & build volume).

Hope the +1 data point helps.

Bupher, 2021-12-24:
I have the same issue. Using prusa slicer 2.4, not on a prusa printer though, and using the print time genius plugin

stevebus, 2022-02-08:
Same issue...   Cura for slicer.  Print Time Genius is installed and working properly

rayge99, 2022-02-14:
I'M Still having this issue. and I'm not sure how to debug the issue.  looking at the log I'm seeing JSON strings that look right to me.. so I'm confused as to why It's still showing 0 on the time left.

Linnie73, 2022-07-04:
Still an issue. Time to filament change is reported at 0:00:00. TTF's log is reporting print time left to be the Octoprint approx. print time of 15.445 seconds or 4:17:25 (see Octo screenshot), not Print Time Genius print time, nor Cura estimated print time in the Gcode file. The gcode contains 3 M600 commands. 
Attached are log files for Print Time Genius, TTF at the start of the print and TTF during the print reporting failures, the gcode in question and screen shots of Octoprint and settings. Hope this helps. 
 
[plugin_TimeToFilament_engine underway.log](https://github.com/eyal0/OctoPrint-TimeToFilament/files/9038420/plugin_TimeToFilament_engine.underway.log)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-TimeToFilament/files/9038421/plugin_PrintTimeGenius_engine.log)
[plugin_TimeToFilament_engine start.log](https://github.com/eyal0/OctoPrint-TimeToFilament/files/9038422/plugin_TimeToFilament_engine.start.log)
[image]
[image]

larp-welt, 2022-08-16:
Still the same problem here. Using:

Octoprint 1.8.2
Python 3.7.3
PrintTimeGenius 2.2.8
TimeToFilament 1.2.1

pyrokidd89, 2022-12-05:
Same issue here also. Is this plugin dead? I have not seen anything from the author in a while.

eyal0, 2022-12-06:
@pyrokidd89 A lot of people seem to have the same problem as you but also a lot of them seem not to.  I have never been able to have the problem on my computer so it's difficult for me to fix it.  If someone that can do some programming has this problem and wants to work on it, I can help out.  But without seeing the problem myself, I'm not sure what to do.

Maybe I log in to some other OctoPrint server that has this problem?  I don't have any ideas.

rayge99, 2022-12-06:
Mine seems to be working on the filament change but random on my own regex for M0. sometimes it works. sometimes it doesn't. 
Msg me and I can get you a login?