eyal0/OctoPrint-TimeToFilament #29: [Feature Request] Display TIme to/of Next Pause

puterboy, 2022-08-26:
Display "Next Pause in" and "Next Pause at" (based on M601 gcode) similar to the current "Next Filament Change in" and "Next Filament change at" (based on M600 gcode).

I know I can presumably add this easily myself by just cutting and pasting the M600 code and substituting M601 but it would nice to have this as a built-in. This is useful if you need to add a magnet or a ball bearing or whatever mid-print...

eyal0, 2022-08-29:
Okay.  Can you check that it works?

go to the plugin installation page of OctoPrint and install from a URL:

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/refs/heads/next_pause.zip

eyal0, 2022-09-01:
Done in #33

closed 2022-09-01 by eyal0