eyal0/OctoPrint-TimeToFilament #30: "Time to Next Layer" not displaying...

puterboy, 2022-08-26:
Unless I am missing something, I don't see the "Time to Next Layer" displaying even though it is checked in the settings.
I am printing with a Prusa MK3S+

[image]

puterboy, 2022-08-26:
OK. I figured out the problem...
Prusa printers use a different comment to signal a layer change --- "^;LAYER_CHANGE"

So use the following settings to get it to work for Prusa printers:

Description: Time to Next Layer
Regex: ^;LAYER_CHANGE
Format: Next layer in <b>${formatDuration(this.progress.printTimeLeft - this.plugins.TimeToFilament["^;LAYER_CHANGE"].timeLeft)}</b>

This may be helpful to add to the documentation...

eyal0, 2022-09-01:
It's not the printer that is using a different format, it's the generator of the stl file.  So like, Cura versus Prusa slicer, etc.

eyal0, 2022-09-01:
If you can send me a sample file and I can adjust the default config to support that other format of layer change.

puterboy, 2022-09-01:
The gcode is ridiculously simple. As per the regex in my previous post, the layer change line simply looks like:
```;LAYER_CHANGE```

Part of the reason I suggested a feature request to track the number of matches as of the current position is so that one could add a layer number to the layer detection since Prusa slicer default layer change comment doesn't include the layer number.

eyal0, 2022-09-02:
Ah, I see! There's no number! Hmm. I'll have to think about how to fix this!

closed 2022-09-02 by eyal0

puterboy, 2022-09-02:
That's why I was thinking it would be helpful to have a general utility that could count instances :)