eyal0/OctoPrint-TimeToFilament #36: After first filament change it always reads 24 hours until the next one

Mat-Hatter, 2022-09-03:
After the first filament change, it always reads 24 hours until the next one. It would be nice to get the time between each one on multiple changes. Is there a setting I am missing for multiple changes?

eyal0, 2022-09-10:
Is 24 hours the wrong answer?

Can you explain what you mean by multiple changes?  As in, what is the expected behavior for you?

Mat-Hatter, 2022-09-10:
Certainly. 

Current behaviour

I add in multiple M600 filament changes to a single print
Next Filament Change shows proper time to first M600
Change Filament and resume print
Next Filament Change Shows 24 hours remaining to Second M600
Change Filament and resume print
Next Filament Change Shows 24 hours remaining to Thrid M600

Expected Behaviour  

I add in multiple M600 filament changes to a single print
Next Filament Change shows proper time to First M600
Change Filament and resume print
Next Filament Change Shows proper time remaining to Second M600
Change Filament and resume print
Next Filament Change Shows proper time remaining to Thrid M600