eyal0/OctoPrint-TimeToFilament #38: This plugin wasn't working for me and others are having the same problem. Solution

tsk-2222, 2023-02-07:
This plugin wasn't working for me and others are having the same problem.

Here is my solution.

The problem (as I see it) is that the plugin always searches the gcode for layer changes and the default regex search expression is configured to only work with certain slicers - simplify3d for example. 

IOW, the plugin is looking for gcode lines at each new layer that look like this: "; layer 1".
Importantly, this is true even when you are only looking for filament changes or pauses.

Furthermore, the setting to fix this is undocumented and hidden under the advanced settings. 

Fix 1: Edit your slicers "layer change custom gcode" section to include something like "; layer {layer_num}"

Fix 2: go to this plugin's advanced section and change the "Time to Next Layer" section so that the regex search expression matches whatever your slicer is outputting. The results can be tested by pasting some gcode into the box near the bottom of the advanced page.

Hopefully future documentation will spell this out or maybe the plugin could be made to handle multiple slicers out of the box like DisplayLayerProgress.

Incidentally, DisplayLayerProgress uses these snippets for different slicers:  

1		[;\s*LAYER:\s*([0-9]+).*]		CURA
1		[; layer ([0-9]+),.*]		Simplify3D
1		[;LAYER:([0-9]+).*]		ideaMaker
count	[;LAYER_CHANGE]		Prusa
count	[; BEGIN_LAYER_OBJECT.*]		KISSlicer
count	[;BEFORE_LAYER_CHANGE]		Slic3r

eyal0, 2023-02-07:
I am not sure if this is the same problem that others were having, but maybe you are right!

If you would like to fix this not just for yourself but for everyone, you can do so by suggesting a change in the code.

On [these lines](https://github.com/eyal0/OctoPrint-TimeToFilament/blob/master/octoprint_TimeToFilament/__init__.py#L32-L35), suggest a change for the format.  Notice that the change will need to be both of the regex and of the format!

eyal0, 2023-02-07:
You can modify that single regex or make a few of them, one for each printer.

charely6, 2023-11-20:
I couldn't figure out how to get the regex that matched the Prusa (orcaslicer) one but I added the comment ; layer [layer_num] to the machine G-Code for Layer change G-code and now it seems to be working.