eyal0/OctoPrint-TimeToFilament #41: Installation fails

Youras, 2023-04-25:
Hello, 

my octopi has no intenet connection, thus I install all the plugins via the webapplication with uploading the zip folder.
However, this plugin seems to require internet connection, since I get some HTTP request error 

`
ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /eyal0/OctoPrint-PrintTimeGenius/archive/master.zip (Caused by NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb547d808>: Failed to establish a new connection: [Errno 111] Connection refused'))`

Can this be resolved somehow?

Best,
Raschid

eyal0, 2023-04-25:
The PrintTimeGenius plugin is required for the TimeToFilament plugin to work. Do you have PTG installed?

Youras, 2023-04-26:
Your first advice did the job, thank you!